openradar/erad2024 #6: Tutorial Idea: Comparison GPM vs Italian/Swiss Ground Radar Network

ghiggi, 2024-05-07:
I started to make a quick search of interesting rainy GPM overpass over Italy since 2019. 
We could select a couple of these events and ask for ground weather radar data from the Italian or Swiss network for analysis and comparison using xradar and wradlib.

All quicklooks can be downloaded [here](https://enacshare.epfl.ch/d6ukhase5iPNtmDXxYzdc).
Here below I showcase some interesting quicklooks (in terms of estimated near surface precipitation and reflectivity at Ku band)

[image]
[image]

[image]
[image]

Genova  
[image]
[image]

[image]
[image]

Naples
[image]
[image]

Piemonte / Lake Maggiore (we could use MeteoSwiss data for these cases)
[image]
[image]

[image]
[image]

[image]
[image]

kmuehlbauer, 2024-06-24:
@ghiggi I'll add an example on Calibration of Ground Radar with GPM.

ghiggi, 2024-06-26:
Hey @aladinor @kmuehlbauer. I started exploring what xradar currently offers to develop a notebook for ground-based radar PPI and RHI comparison with GPM ... and then dig a bit further with wradlib utilities.
I was thinking to develop tutorials using the Columbian Radar Network, NEXRAD and the Finnish network (via AWS). 
I might have some questions to address you on what are you planning to do in the coming weeks in terms of tutorials and what is in the roadmap in term of xradar accessor methods for `xr.Dataset` and `xr.Datatree` ... 

Can we maybe set up a meeting in the coming days?

kmuehlbauer, 2024-06-26:
@ghiggi 

Yeah, going idea to coordinate. 

What I had in mind is something along this notebook from our lecture series (unrendered version, single PPI matching):

https://nbviewer.org/urls/git2.meteo.uni-bonn.de/projects/radarmeteorology/repository/revisions/SS2024/raw/notebooks/Lesson_07-Satellite_Calibration/Satellite_Calibration.ipynb.raw

I'd update the notebook to use gpm-api for loading. The matching is based on wradlib under the hood.

aladinor, 2024-06-26:
Hi @kmuehlbauer and @ghiggi,

Thank you for bringing this up. I think we can use one of these examples to create a workflow using `xr.Datatree`. We could structure a datatree with all the radar/GPM data in our example and then export it to Zarr. I have a [poster](https://github.com/aladinor/raw2zarr/blob/main/notebooks/Poster.ipynb) for ERAD that demonstrates the advantages of using datatree.

Additionally, @mgrover1 mentioned that we can host the ARCO files on the Pythia S3 Object storage. This would ensure a smooth workflow during the short course.

Please let me know your thoughts.

mgrover1, 2024-06-26:
Agreed - let's host the zarr store on Pythia + include both in this short course, as well as the Pythia cookbook.

Here is a link to an example of accessing data on the bucket!
https://projectpythia.org/lasso-those-clouds-cookbook/notebooks/LassoARMClouds.html#access-lasso-sgp-data-from-the-nsf-jetstream-cloud

Happy to help here!

ghiggi, 2024-07-10:
Hi @kmuehlbauer ! 
I start to go through the notebook you mentioned but I coudn't find the `radarmeteorology.volume_matching` function in the repository. Could you provide it to me?

kmuehlbauer, 2024-07-10:
@ghiggi I'm currently traveling and may not respond in due time.

AFAIK @vlouf's code is numpy based. It would be great to stay in the xarray world, if possible. 

The code you are missing is located in the same repository in the radarmeteorology subfolder:

https://git2.meteo.uni-bonn.de/projects/radarmeteorology/repository/revisions/SS2024/entry/radarmeteorology/radarmet.py

You might just copy over anything you need from there. I've planned to move this into wradlib later this year. Let me know how things go. :heart:

ghiggi, 2024-08-22:
The documentation for the project are currently available on the GPM-API documentation at: 
- [Spaceborne-Ground Radar Matching (Methodology)](https://gpm-api.readthedocs.io/en/latest/tutorials/tutorial_03_SR_GR_Matching.html)
- [Spaceborne-Ground Radar Calibration (Application)](https://gpm-api.readthedocs.io/en/latest/tutorials/tutorial_03_SR_GR_Calibration.html)

We should discuss if we want to run the afternoon project on Binder (providing example GPM and xradar datasets) or rather let the students set up the environment on their laptop and analyze their own ground radar data of interest, with me providing the GPM overpass time required for the analysis.

kmuehlbauer, 2024-08-26:
@ghiggi We should definitely have one example to run on binder. In general every participant will use it's own binder instance to run the notebooks. It would be great to merge this asap.

Regarding the Italian data, I've contacted Gianfranco Vulpiani to get some data for a recent overpass on 2024-05-21 06:30 UTC but did not hear back until now.

aladinor, 2024-09-03:
Hi @ghiggi, 

Data from 8 Colombian radars are currently available in this Amazon s3 [bucket](https://registry.opendata.aws/ideam-radares/). It is open data but not FAIR data. If you can point me to some GPM overpasses I can get the data for you, or you can directly get it from there. 

Let me know your thoughts.

ghiggi, 2024-09-03:
Yes I saw the data on S3 and I know how to retrieve it. If I find the time I will prepare the list of GPM overpass for ERAD. Otherwise we can assess the calibration after ERAD if you are interested ;)

aladinor, 2024-09-03:
Totally interested. Let me know how can I help

kmuehlbauer, 2024-09-05:
Great idea to follow up at or after ERAD, otherwise this seems very much complete.

closed 2024-09-05 by kmuehlbauer