openradar/erad2024 #7: Add reader for italian radar data data provided by Gianfranco Vulpiano

wolfidan, 2024-05-08:
Hi,
Jordi asked Gianfranco Vulpiani for a usable dataset for the ERAD short course and he provided this:

https://zenodo.org/records/4897697

The data uses a proprietary format called Datamet (https://zenodo.org/records/4897245). We would need to write a reader to ingest this data for the course? I propose to write the reader. Should I write it in xradar or pyart?

Thanks,
Cheers,
Daniel

kmuehlbauer, 2024-05-08:
@wolfidan xradar would be preferred. : grinning:

wolfidan, 2024-06-26:
Thank you @kmuehlbauer , I'm on it now, I'll keep you posted.

wolfidan, 2024-06-27:
Hi @kmuehlbauer ,

I made a first attempt here: https://github.com/wolfidan/xradar/blob/main/xradar/io/backends/datamet.py

However I am not sure how to proceed to make it available in xarray? 

The first part of this code works (up to xr.open_dataset):

`f = '/scratch/wolfensb/erad2024/datamet_reader/H-000-VOL-ILMONTE-201907100700.tar'
import xarray as xr
import xradar as xd
entry = xd.io.backends.DataMetEntrypoint()
ds = entry.open_dataset(f, group='sweep_1')
xr.open_dataset(f, group='sweep_1', engine = 'datamet')`

But the xarray open_dataset fails with

`ValueError: unrecognized engine datamet must be one of: ['netcdf4', 'h5netcdf', 'scipy', 'cfgrib', 'cfradial1', 'furuno', 'gamic', 'gini', 'iris', 'nexradlevel2', 'odim', 'radolan', 'rainbow', 'store', 'wradlib-cfradial1', 'wradlib-cfradial2', 'wradlib-furuno', 'wradlib-gamic', 'wradlib-iris', 'wradlib-odim', 'wradlib-rainbow']`

Is there something else I need to do to make it available?
Note: xradar (with my modifs) was installed with python -m pip install .

Thanks a lot

kmuehlbauer, 2024-06-27:
@wolfidan

Great! 

This should work, too, for your testing:

```python
xr.open_dataset(f, group='sweep_1', engine=xd.io.backends.DataMetEntrypoint)
```

kmuehlbauer, 2024-06-27:
@wolfidan If we are allowed to put one testfile on [open-radar-data](https://github.com/openradar/open-radar-data) then we could use that for testing and example notebook. If you think this is ready for a PR please go for it.

wolfidan, 2024-06-27:
@kmuehlbauer 
Data is publicly available so I think it is ok to upload one file.
Thanks for your help !
Not yet ready for PR, I need to write tests and do additional checks, but I'll try to make it quick

ghiggi, 2024-08-23:
Solved by https://github.com/openradar/erad2024/issues/7

closed 2024-08-23 by ghiggi