openradar/erad2024 #10: Course Schedule

kmuehlbauer, 2024-06-24:
!! Preliminary !!

**Morning**

- Open Science overview (30 - 45 min) (must)
- Toolkit basics - xradar/command line tools/xarray, LROSE (GUI, CLI), BALTRAD (1 to 1.5 hours)
    - Research (30 minutes)
        - xradar
        - Py-ART/wradlib
        - LROSE
        - Visualization (tools, colormaps, …)
    - Operations (30 minutes)
        - Pyrad
        - LROSE
        - BALTRAD
    - Sample workflows
        - Xradar → Py-ART + wradlib
        - GPM-API + wradlib
        - LROSE wind tool?
        - BALTRAD GUI
    

What aspects of the radar processing chain where each package is useful; make a flowchart of data flow?  
See this for example: https://github.com/openradar/asean2024-pyrad-course/blob/main/presentations/pdf/01_open_meteorological_radar_software_constellation.pdf

**Afternoon**

- Some set of curated datasets
- GPM - ground based radar comparison
- NEXRAD - potential for multidoppler
- xarray/datatree/zarr + dask (cloud/arco format) Colombian Sigmet radar data - ARCO! 
- OPERA data? Focused on Italy?

jfigui, 2024-07-19:
@openradar/erad2024-team 

I am looking at the Preliminary time schedule. With @wolfidan we were planning a general 30 min (could be reduced to 20 min) introduction (in the morning or early afternoon) and a group work focused on QPE.  Could we set the 20 min slot right after the Py-ART/wradlib intro? We do not need a specific slot in the afternoon then.

kmuehlbauer, 2024-08-23:
Resolved by #83

closed 2024-08-23 by kmuehlbauer