openradar/erad2024 #11: Intro to LROSE

leavesntwigs, 2024-06-24:
Create a notebook for intro to LROSE
* intro
* visualization
* wind
* operational with LROSE

---

afternoon workflows (interactive)

kmuehlbauer, 2024-06-25:
@leavesntwigs LROSE is installed and available via the binder link. Please check, if everything is good.

kmuehlbauer, 2024-06-25:
@leavesntwigs So it seems the conda python env takes precedence.

I've got LROSE (RadX examples from http://wiki.lrose.net/index.php/Lrose_quickstart#Install) to work by overriding the env PATH to point to the lrose binaries and the system hdf5/netcdf-c binaries. @mgrover1 Maybe you have some idea how to properly fix this for LROSE?

I've already checked, we can't use hdf5 1.10.7 in the conda env.

leavesntwigs, 2024-06-27:
I launched the binder link and tried some quick lrose commands in a terminal window.  This all worked well. Thank you!
I then tried the same test commands from a notebook and things looked good, but I also see the hdf5 error messages.  The hdf5 error messages only appear when /usr/local/lrose/bin/RadxPrint -h from a notebook.  No error in a terminal.

kmuehlbauer, 2024-08-13:
merged with #34

closed 2024-08-13 by kmuehlbauer