openradar/erad2024 #40: Invalid response 413 request entity too large

kmuehlbauer, 2024-07-04:
Some things persist somehow.

When the notebook size exceeds a certain limit saving will generate an error, like this:

`Invalid response 413 request entity too large`

See also https://github.com/openradar/erad2022/issues/69.

A quick search connects this error to some problematic/missing configuration for nginx or whatever serve is used in the main jupyter docker container.

- https://discourse.jupyter.org/t/invalid-response-413-request-entity-too-large/13531
- https://github.com/jupyterlab/jupyterlab/issues/4214

Not sure, if we can do anything about it.

kmuehlbauer, 2024-08-12:
@openradar/erad2024-team Is this still an issue?

kmuehlbauer, 2024-08-13:
Obviously, yes:

[image]

kmuehlbauer, 2024-08-14:
XRef: https://discourse.jupyter.org/t/invalid-response-413-request-entity-too-large/13531/2

@mgrover1 Would be nice, if you could forward this to the pythia people. Thanks!

mgrover1, 2024-08-14:
@kmuehlbauer - I am checking to see if we can set:

```bash
client_max_body_size 0
```

Which based on that thread, should resolve the issue.

kmuehlbauer, 2024-08-14:
@mgrover1 That would have to be set on the pythia infrastructure AFAICT.

ktyle, 2024-08-14:
@mgrover1 @kmuehlbauer I'm looking into it on our Pythia binder!

kmuehlbauer, 2024-08-14:
@ktyle Thanks! If you fear to much load, you might experiment with increasing client_max_body_size. The issue happens when the notebook becomes to large due to huge images rendered therein.

ktyle, 2024-08-14:
@zonca how might I configure my JS2 binderhub to configure the nginx ingress as above?

zonca, 2024-08-14:
here is for jupyterhub:
https://github.com/jupyterhub/zero-to-jupyterhub-k8s/issues/1230

It should be the same or fairly similar in Binderhub:
https://binderhub.readthedocs.io/en/latest/https.html

zonca, 2024-08-14:
so add:

        nginx.ingress.kubernetes.io/proxy-body-size: 64m

under `annotations` of your NGINX configuration in the binderhub yaml file

ktyle, 2024-08-15:
Thanks, @zonca ! I have added that line to the configuration file and re-deployed. @kmuehlbauer @mgrover1 Can you give it another try and see if the 413 errors are no longer occurring? I just ran through the 2_BALTRAD_DRQC notebook and so far, no 413's.

kmuehlbauer, 2024-08-16:
@ktyle @zonca Thanks for taking care :heart:! The changes resolved the issue :tada:!

closed 2024-08-16 by kmuehlbauer