openradar/erad2024 #51: QPE exercise in the North of Italy

jfigui, 2024-08-07:
Discussion of all topics related to the QPE exercise.

jfigui, 2024-08-07:
@openradar/developers @wolfidan ,

I opened this topic to discuss the QPE exercise. I understood from our meeting on the 5th of August that the topic will be driven by Pyrad but other software packages can contribute with notebooks as post-course exercises.

Could we maybe start by identifying the contributors to this project? Please add your name and contribution as a reply to this comment:
- @wolfidan , @jfigui : Development of a data processing chain with Pyrad config files. (main material during the course)

jfigui, 2024-08-07:
@openradar/developers @wolfidan Let me know if this brief description of the project suits you:

In this project we will learn how to setup a data processing chain for radar rainfall rate retrieval. The project will show the main steps involved between reading raw polarimetric moments to comparing rainfall rate retrievals with rain gauges, including clutter filtering, PhiDP estimation, KDP retrieval, etc. The processing chain implementation will be performed with [Pyrad](https://github.com/meteoswiss/pyrad).

kmuehlbauer, 2024-08-20:
@jfigui @wolfidan Can you please give some status update? Are you about to have a PR ready for the notebook?

wolfidan, 2024-08-20:
@kmuehlbauer ,

Sorry for the delay, I had quite a bit of cleaning to do in pyrad and Jordi is in vacation this week. I am on it, I should be able to have a PR this week hopefully.

kmuehlbauer, 2024-08-20:
Thanks @wolfidan, great to hear it's progressing. :+1:

wolfidan, 2024-08-27:
I want to do the PR, but have an issue. How should I manage it with the data? The thing is that the QPE exercice processes a few hours of data (otherwise it does not make much sense). So if I put the whole dataset on git, it's 2.2 Gb...How should I proceed? We don't have a way in pyrad at the moment to download data from an S3 (only to upload the output products). A possibility would be to have the dataset on a S3 and download it with an additional scripts during creation of the binder?

kmuehlbauer, 2024-08-27:
@wolfidan Which data are we talking about? This data: 20240522_MeteoSwiss_ARPA_Lombardia? This is already available in the pythia s3 bucket. This can be accessed by fsspec using local scheme. So should work for pyrad.

If you talking about another dataset, please link the data here so that @mgrover1 can add it to the pythia s3 bucket.

aladinor, 2024-08-27:
He @wolfidan and @jfigui, as I mentioned in #94, we have created an Analysis-ready dataset. We can use it for QPE and QVPs. This is the workflow I suggest with the Colombian [radars](https://nbviewer.org/github/aladinor/notebooks/blob/557524a169662255859d4011fca56a5923cdc3df/AtmosCol-FAIR-radar.ipynb). 

Let me know if we can merge at some point what you have in mind with this dataset

jfigui, 2024-09-03:
@aladinor ,

We initially planned for a fully pyrad QPE exercise. What do you think if we let you some minutes to introduce your workflow at the beginning of the exercise as an add on?

aladinor, 2024-09-03:
Hi @jfigui,

Thank you for the suggestion. The QPE I’ve planned for the Analysis-Ready sections primarily demonstrates the use of a theoretical Z-R relationship. My goal is to showcase how efficiently we can leverage the ARCO dataset for certain operations, rather than diving deep into advanced QPE methodologies.

I understand that this approach might be simpler compared to the more comprehensive PyRad exercise you’re planning. If you'd like, we could explore how to integrate these approaches or adjust the scope to better align with your plans. I’m happy to discuss further, and we can also review the [QPE exercise](https://openradarscience.org/erad2024/qpe-qvps) together to find the best way forward.

kmuehlbauer, 2024-09-05:
Given the time left until course this seems complete,too.

closed 2024-09-05 by kmuehlbauer