openradar/erad2024 #53: Creating analysis-ready data with xradar with QVPs (xradar)

jfigui, 2024-08-07:
Discussion of all topics related to QVPs

jfigui, 2024-08-07:
@openradar/erad2024-team , @wolfidan 

I took the liberty to open this issue to discuss the QVP exercise. Could we start by identifying the contributors to this project? Please add your name and contribution as a reply to this comment:
- @wolfidan , @jfigui : We will provide Pyrad config files to plot QVPs

aladinor, 2024-08-07:
Thanks, Jordi, for creating this issue. I will work on analysis-ready data; one application could be QVPs or QPE. The idea is to show the advantages of having data in CFRadial2/FM301.

aladinor, 2024-08-08:
By the way, @mgrover1 radar data has already been converted into ARCO format. How can I upload the data to the bucket?

aladinor, 2024-08-12:
Analysis-ready data could be conditioned to this [issue](https://github.com/pydata/xarray/pull/9198) since we might need to pass credentials. Let me know your thoughts @kmuehlbauer @mgrover1

mgrover1, 2024-08-13:
@aladinor - the zarr file is now available on the bucket!

```bash
https://js2.jetstream-cloud.org:8001/pythia/radar/erad2024/zarr_radar/erad_2024.zarr
```

kmuehlbauer, 2024-09-05:
AFAICT this is up and running.

closed 2024-09-05 by kmuehlbauer