openradar/erad2024 #87: Keep Jupyter Notebook outputs

ghiggi, 2024-08-23:
Currently the `nbstripout` precommit remove outputs and figures from the Jupyter Notebooks. 
Do we want to maybe add `--keep-output` to the precommit arguments or there is a reason to avoid that? File size? 
Currently we need to `git commit <message> -n` to avoid removal of the outputs if the precommit is installed.

I can create a PR if needed!

kmuehlbauer, 2024-08-23:
The notebooks are kept without outputs in git. In order to not check in any outputs, this is stripped from the notebok.

The notebooks are rendered in CI and put on gh-pages.

kmuehlbauer, 2024-08-23:
Later we might used the markdown versions of notebooks inside the repo to reduce the overhead and make PR review easier. And with later I mean not before ERAD.

closed 2024-08-23 by ghiggi