openradar/erad2024 #92: coexistence of ARM and MeteoSwiss Py-ART

jfigui, 2024-08-26:


jfigui, 2024-08-26:
@openradar/erad2024-team ,

We have an issue with the coexistence of the ARM and the MeteoSwiss versions of Py-ART. Some features that we use in our QPE exercise are in the MeteoSwiss version of Py-ART but not in the ARM version. We need your advise on how to proceed:
- Create two separate environments, one containing the MeteoSwiss version and one the ARM version
- Add the missing MeteoSwiss features to the ARM Py-ART
- Use the MeteoSwiss Py-ART instead of the ARM Py-ART in the course (all ARM features are available in the MeteoSwiss version)
- Other ...

Please let us know your preferred solution.

kmuehlbauer, 2024-08-26:
@jfigui Maybe the quickest solution would be to add a second conda-environment. I've added it to todays meeting agenda for discussion.

wolfidan, 2024-08-26:
@jfigui @kmuehlbauer 
One quick and dirty solution I used at last ERAD was to simply run 
pip install pyart_mch before running our part of the course.
I don't know if it's a viable solution but since our part would be in a parallel session it could maybe work.

kmuehlbauer, 2024-08-26:
> One quick and dirty solution I used at last ERAD was to simply run
> pip install pyart_mch before running our part of the course.

Let's have this as our last resort. It's hacky and we should really try to come up with a proper solution for the course.

1. The cleanest solution would be to have the changes upstreamed to arm_pyart. 
2. Another option would be to create a second environment.
3. Hacky solution by overinstalling pyart_mch on-site.

If 1. is not manageable, we should aim for 2.

mgrover1, 2024-08-27:
Can we put together an issue on the Py-ART side highlighting what is needed for 1:1 functionality for the short course? A list of modules/submodules?

wolfidan, 2024-08-27:
Hi @mgrover1 ,
There are not so many features to add, I'm working on the PR at the moment, and making the unit tests for the features I added. Hopefully I can do the PR tomorrow.

mgrover1, 2024-08-27:
wonderful! Please let me know how we can assist in getting this merged 👍

mgrover1, 2024-08-30:
With the release of Py-ART 1.19.0, this should be resolved; let's wait for the conda-forge wheels to be released and try this :)

kmuehlbauer, 2024-09-04:
Currently (#100) `pyrad_mch` and ``pyrad_arm` are installed into conda. Is this correct? If there is any possibility for clashes, one should be removed (pyrad_arm I suppose)?

mgrover1, 2024-09-04:
Any issue with just using `pyrad_arm` @wolfidan @jfigui ? Now that those features are included in the latest release of ARM Py-ART

wolfidan, 2024-09-04:
@mgrover1 , pyrad_arm is fine. I just forgot to remove pyrad_mch from the environment.yml. Sorry about this. Please see https://github.com/openradar/erad2024/pull/100

kmuehlbauer, 2024-09-05:
Please reopen, if this isn't finished yet.

wolfidan, 2024-09-05:
Hi @kmuehlbauer @mgrover1 
Just noticed that in the binder it is still the pyart version 1.18.7 that gets installed.
Could we please force 1.19, it would just be a matter of adapting environment.yml in the binder

kmuehlbauer, 2024-09-05:
Wrong issue reported, just for reference. pyart 1.19 is still pending pr in conda-forge. Would be great, if this can be merged asap, @mgrover1 @zssherman. Thanks.

mgrover1, 2024-09-05:
Merged 👍

kmuehlbauer, 2024-09-05:
Great, let's wait another while an restart #104.

closed 2024-09-05 by kmuehlbauer