openradar/erad2024 #98: xarray v2024.7.1. version

aladinor, 2024-08-30:
Hi @kmuehlbauer and @mgrover1. As we already mentioned [here](https://github.com/openradar/erad2024/issues/94#issuecomment-2315335713), Analysis-Ready and GPM notebooks require xarray v2024.7.1... development version.  How can we include it in the docker/Pythia binder?

Ideally, we would like xarray from the GitHub repo. Something like 

```shell
python -m pip install git+https://github.com/pydata/xarray.git
```

Please let me know your thoughts.

mgrover1, 2024-08-30:
@kmuehlbauer - do you know if there are plans for an xarray release soon?

kmuehlbauer, 2024-08-30:
@aladinor This would have to be added in the appendix.txt in the binder folder.

kmuehlbauer, 2024-08-30:
@mgrover1 No, I've missed this month's dev meetings. But there is no indication in the meeting notes. We can't depend on release which might not land in time. So best to add it to the docker image.

mgrover1, 2024-08-30:
We can add that in the pip dependencies I believe?

kmuehlbauer, 2024-08-30:
Let's try it like that, if that doesn't work, we can over install in appendix.txt.

mgrover1, 2024-08-30:
@aladinor - just merged the PR, give it a shot when you get a chance 😄

closed 2024-08-30 by mgrover1

aladinor, 2024-08-30:
Great! Thanks @mgrover1

PR #99 (2024-08-30): FIX: Use the most recent commit of xarray main
Closes #98 

Use a specific branch of xarray github main branch