openradar/erad2024 #112: Add Italian GPM data + workflow

kmuehlbauer, 2024-09-16:
We did not make it before ERAD2024, but I've received 3 volumes of italian radar data to extend our GPM Matching/Calibration.

@mgrover1  I'll check the data the next days and will contact you to put it on the pythia bucket.
@ghiggi It would be great if you could extend/add the processing for this data, once it's available.

ghiggi, 2024-09-16:
Yes. I can do it as soon as the data are available. I can also provide a list of all GPM overpass over Italy for who want to extend the analysis. 
With @aladinor we might provide a tutorial for the Columbian radars too ;)

kmuehlbauer, 2024-09-16:
> Which xradar reader should be used to read the Italian data?

It's in the datamet-format which @wolfidan implemented a reader just before ERAD2024.

```python
import xradar as xd
fname = "somename.tar.gz"
tree = xd.io.open_datamet_datatree(fname)
```

We'll have to wait until @mgrover1 is back in office. But you can already play with the data. Thanks @ghiggi.

ghiggi, 2024-09-17:
@kmuehlbauer I had a quicklook this morning, but there is something strange in the GR data. 
This is the GPM DPR surface reflectivity measured at 06:30 UTC. The cross indicates the position of the GR
[image]

If I look at the GR data, I see something that correlates to the shape of the storm at sweep higher than 6, but with reflectivities below 10 dBZ. 
 
[image]
[image]

[image]
[image]

[image]
[image]

There is something going wrong with the decoding/scaling of the GR data I think @wolfidan ! If the `scale_factor` is defined as `1/current_scale_factor` in the reader I obtain this which looks more realistic to me ;) 

[image]

ghiggi, 2024-09-17:
@kmuehlbauer for the tutorial we will need to filter out the clutter and some additional preprocessing of GR data before doing the comparison. Do we use pyart or wradlib? Do you have already some default/template processing to use?

kmuehlbauer, 2024-09-17:
@ghiggi The radar obviously operates in top-down mode. So higher elevations are scanned first.

You are referring to these [lines of code](https://github.com/openradar/xradar/blob/34a9e8234189f921b1586001c84bdaf8f80f4c48/xradar/io/backends/datamet.py#L166-L167)?

I'm not sure what the correct setting of `offset` and `slope` is, but if I use a negated offset (and keep the scale_factor), the output looks even more realistic. Can you confirm that @ghiggi?

For preprocessing examples I usually use clutter detection from wradlib to remove clutter and make some interpolation afterwards to fill the holes. But we can also use pyart for preprocessing. I've no fixed preprocessor pipeline, always build it from scratch for new radars.

ghiggi, 2024-09-17:
I assumed C band radar, empirical correction of Tan for Ku to C, beamwidth of 1° ... plot some stuffs and I don't think it's a good overpass for assessing GR bias. Most volumes are above 35 dBZ, ground clutter, beam blockage, NUBF, ... I played a bit with filtering criteria but ...

[image]
[image]
 
[image]

ghiggi, 2024-09-17:
@kmuehlbauer I was looking more at [these lines](https://github.com/openradar/xradar/blob/34a9e8234189f921b1586001c84bdaf8f80f4c48/xradar/io/backends/datamet.py#L281).
I see `scale_factor` and `add_offset` added to the `attrs`, but I found these in the `encoding` once I open the datatree. This means xarray/cf decoding is going on moving from `attrs` to `encoding`. Are we applies scaling twice? I can't have a look at it for the rest of the day ... I might be able to give a look tomorrow ...

kmuehlbauer, 2024-09-17:
@ghiggi 

You can deal with this the following:

```python
import xradar as xd
swp = xr.open_dataset(fname, engine="datamet", group="sweep_11", decode_cf=False)
swp.DBZH.attrs["add_offset"] = swp.DBZH.attrs["add_offset"] * -1
swp = xr.decode_cf(swp)
```

`add_offset` and `scale_factor` are moved to the attributes to make them available for automatic decoding in xarray. This is what happens, but it looks like the offset needs to be the other way round.

Take your time, no rush.

ghiggi, 2024-09-17:
Quickly tried and this is not correct:
```
import xarray as xr
swp = xr.open_dataset(fname, engine="datamet", group="sweep_11", decode_cf=False)
swp.DBZH.attrs["add_offset"] = swp.DBZH.attrs["add_offset"] * -1
swp = xr.decode_cf(swp)
```
If we open with `decode_cf=False` we get directly: 
[image]
If we do 
```
    ds_gr["DBZH"].attrs["scale_factor"] = 1/ds_gr["DBZH"].attrs["scale_factor"] 
    ds_gr = xr.decode_cf(ds_gr)
```
we get this: 

[image]

kmuehlbauer, 2024-09-17:
Thanks @ghiggi, I'll have a deeper look later today.

kmuehlbauer, 2024-09-17:
True @ghiggi, your hunch is correct! :clap: scale offset is applied twice, once in the reader and second time in xarray decoding.

https://github.com/openradar/xradar/blob/34a9e8234189f921b1586001c84bdaf8f80f4c48/xradar/io/backends/datamet.py#L182-L183

kmuehlbauer, 2024-09-17:
@ghiggi FYI: https://github.com/openradar/xradar/pull/209

mgrover1, 2024-09-19:
@kmuehlbauer - checking in here now that I am back in Chicago! Is there a link to the data to upload?

kmuehlbauer, 2024-09-19:
ftp://ftp.meteo.uni-bonn.de/pub/radar/erad2024/gpm/vol-serano-202405210625-tar-gz_2024-09-16_0743/H-000-VOL-SERANO-202405210625.tar.gz

ftp://ftp.meteo.uni-bonn.de/pub/radar/erad2024/gpm/vol-serano-202405210625-tar-gz_2024-09-16_0743/H-000-VOL-SERANO-202405210630.tar.gz

ftp://ftp.meteo.uni-bonn.de/pub/radar/erad2024/gpm/vol-serano-202405210625-tar-gz_2024-09-16_0743/H-000-VOL-SERANO-202405210635.tar.gz

mgrover1, 2024-09-19:
@ghiggi @kmuehlbauer - the data is now available in the pythia bucket under the /radar/erad2024/gpm_api directory:
- H-000-VOL-SERANO-202405210625.tar.gz
- H-000-VOL-SERANO-202405210630.tar.gz
- H-000-VOL-SERANO-202405210635.tar.gz

kmuehlbauer, 2024-09-20:
@ghiggi I've released xradar v0.6.5 with the needed fix. When using the binder, we`d have to create an new PR to re-create the docker image with latest xradar.

ghiggi, 2024-09-27:
Hi @kmuehlbauer ! I unfortunately didn't had time to advance on this. Next week I will be in Wurzburg at the EUMETSAT satellite conference ... so I will try to finalize this in two weeks ...

ghiggi, 2025-03-11:
Hi @kmuehlbauer @mgrover1 ! I completely forgot about answering to this issue. 
When I went through the data, I discovered there was few matching footprints (after quality filtering) to make a meaningful comparison. 
In the past months I improved the volume matching routine and now I have scripts to assess the calibration of hundreds of ground radars during the period (2018-2024) through [radar_api](https://github.com/ghiggi/radar_api). If Gianfranco is interested, we could discuss about run it also for the Italian radars. 
BTW Kai I am coming to Bonn next week, so we might discuss further of it in person ;)

mgrover1, 2025-03-11:
Great progress here @ghiggi - I really like all the progress you have made with radar_api 👍